ricosjp/truck #2: STEP I/O

ytanimura, 2020-12-28:
As long as we claim this crates a CAD Kernel, STEP I/O is unavoidable.  
We're working on a plan for this, which we'll announce here soon.

g-rauhoeft, 2021-02-26:
I'm currently working on an EXPRESS to Rust transpiler and STEP parser with the aim of extracting structural and topological information. I was planning on also implementing a basic CAD Kernel to achieve this, until I stumbled across your amazing project. Perhaps we can collaborate on this.

ytanimura, 2021-02-27:
@g-rauhoeft 
Thank you for your kindly suggestion. Now, we are also working on an EXPRESS to Rust transpiler and STEP parser. That is still WIP, but we have decided to make the code public in response to your suggestion.
https://github.com/ricosjp/ruststep
This one is purely a transpiler and parser, so it is maintained in a separate repository. Most of the expansion of truck for STEP will be done pending completion of ruststep. We are waiting for your advice, issues or pull requests for both truck and ruststep!

ytanimura, 2021-02-27:
Of course, since truck and ruststep are independent of each other, truck can support several STEP utilities with different backends.

J-F-Liu, 2021-05-12:
I also did some work on reading STEP files. https://github.com/J-F-Liu/iso-10303

ytanimura, 2021-10-02:
[ruststep](https://github.com/ricosjp/ruststep) has been released!

g-rauhoeft, 2021-10-02:
> [ruststep](https://github.com/ricosjp/ruststep) has been released!

Congratulations and thank you! It's already been a pleasure to work with it and I'm looking forward to contribute more in future. :)

g-rauhoeft, 2022-06-27:
I'm currently working on STEP importing. Is it ok for you if I make a pull request once I'm done or am I interfering with your work/timeline?