ricosjp/truck #5: RSweep with a fixed point

ytanimura, 2020-12-28:
We are not yet prepared for rotational sweeps with a fixed point. For example, we cannot create even a sphere by simple rotational sweeps.  
The biggest difficulty in implementation is that when the surface is created in a naive way, the NURBS surface degenerates around the fixed point and the normal vector at the fixed point cannot be calculated.

ytanimura, 2021-02-16:
We should create a new enum that enumerates special shapes such as planes and spheres in parallel with NURBS, and add rotators to it.

ytanimura, 2021-04-14:
In the master branch, we found holes in the cone and gaps in the mesh at the edges. These will be dealt with in due course, but priority will be given to the STEP output first.

phkahler, 2022-01-30:
> The biggest difficulty in implementation is that when the surface is created in a naive way, the NURBS surface degenerates around the fixed point and the normal vector at the fixed point cannot be calculated.

I don't think you get much choice in the case of a cone, or any other time a curve ends on the axis of revolution at a non-orthogonal angle. We had the degenerate surface/edge/normal come up in Sovlespace. The discussion around the pull request to fix it is [here](https://github.com/solvespace/solvespace/pull/736)

commit d8d68b8d (2024-02-09): Merge branch '5-split-closed-surface' into 'master'

Resolve "shape healing"

Closes #5

See merge request ricos/truck/truck!399
closed 2024-02-09 by ytanimura