ricosjp/truck #7: What's the difference between this project and occ

happydpc, 2021-01-27:
I am watching this project. It's very good that this's implemented in rust. And I wonder what's the major difference between this and opencascade, is there any better design inside this project?

ytanimura, 2021-01-27:
The most significant difference lies in the small-scale crating. Overall, it is a competitor to OCCT, but the truck-platform by itself, for example, is aimed at replacing three.js in the WebGPU generation, or a primitive Unity that uses crates.io as an asset store. This design allows experts to replace individual implementations.. Moreover, it can easily accommodate to algorithm alternatives such as GPU accelerators for NURBS meshing, or unpredictable expansion for unknown new technologies. This is possible because of centralized package management by Cargo.

happydpc, 2021-01-28:
Is there any plan to support boolean operations? Glad to implement these.

ytanimura, 2021-01-28:
Yeah, we also consider that boolean operations are necessary and is the next plan after STEP handling.
We cannot make any official promises about deadlines, however, I'm hoping to release something by August.

g-rauhoeft, 2021-02-22:
Another very important difference is that this project is Apache 2.0 licensed, while OCC is LGPL 2.1.

ytanimura, 2021-08-25:
Close this issue. Boolean operations will be discussed in #17 .

closed 2021-08-25 by ytanimura