ricosjp/truck #9: Failed to create pipeline in rotate-objects example on the DX12 backend

daidanretsu, 2021-02-19:
Hello,
I am having a issue when running the rotate-objects example on the DX12 backend.
```
cargo run --example rotate-objects

thread 'main' panicked at 'failed to create graphics pipeline: Failed to create pipeline: Unsupported usage: Implementation specific error occurred', C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-core-0.6.5\src\device\mod.rs:2898:30
```

Here is the backtrace:
```
stack backtrace:
   0: std::panicking::begin_panic_handler
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b\/library\std\src\panicking.rs:493
   1: std::panicking::begin_panic_fmt
             at /rustc/cb75ad5db02783e8b0222fee363c5f63f7e2cf5b\/library\std\src\panicking.rs:435
   2: wgpu_core::device::{{impl}}::device_create_render_pipeline::{{closure}}<wgpu_core::hub::IdentityManagerFactory,gfx_backend_dx12::Backend>
             at C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-core-0.6.5\src\device\mod.rs:2898
   3: core::result::Result<gfx_backend_dx12::resource::GraphicsPipeline, gfx_hal::pso::CreationError>::map_err<gfx_backend_dx12::resource::GraphicsPipeline,gfx_hal::pso::CreationError,wgpu_core::device::DeviceError,closure-10>
             at C:\Users\user\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\result.rs:595
   4: wgpu_core::hub::Global<wgpu_core::hub::IdentityManagerFactory>::device_create_render_pipeline<wgpu_core::hub::IdentityManagerFactory,gfx_backend_dx12::Backend>
             at C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-core-0.6.5\src\device\mod.rs:2893
   5: wgpu::backend::direct::{{impl}}::device_create_render_pipeline
             at C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-0.6.2\src\backend\direct.rs:758
   6: wgpu::Device::create_render_pipeline
             at C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\wgpu-0.6.2\src\lib.rs:1475
   7: truck_rendimpl::InstanceState::pipeline_with_shader
             at .\truck-rendimpl\src\instdesc.rs:186
   8: truck_rendimpl::PolygonInstance::pipeline_with_shader
             at .\truck-rendimpl\src\polyrend.rs:142
   9: truck_rendimpl::polyrend::{{impl}}::pipeline
             at .\truck-rendimpl\src\polyrend.rs:189
  10: truck_platform::Rendered::render_object<truck_rendimpl::PolygonInstance>
             at .\truck-platform\src\lib.rs:345
  11: truck_platform::Scene::add_object
             at .\truck-platform\src\scene.rs:357
  12: rotate_objects::MyRender::load_obj<slice<u8>>
             at .\truck-rendimpl\examples\rotate-objects.rs:63
  13: rotate_objects::{{impl}}::init
             at .\truck-rendimpl\examples\rotate-objects.rs:136
  14: rotate_objects::app::App::run<rotate_objects::MyRender>
             at .\truck-rendimpl\examples\app.rs:93
  15: rotate_objects::main
             at .\truck-rendimpl\examples\rotate-objects.rs:282
  16: core::ops::function::FnOnce::call_once<fn(),tuple<>>
             at C:\Users\user\.rustup\toolchains\stable-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core\src\ops\function.rs:227
```

I found an similar issue reported https://github.com/gfx-rs/wgpu-rs/issues/600.

Thanks.

ytanimura, 2021-02-19:
Thank you for your kind issue.

I was able to reproduce the problem in my environment. I've even identified the cause of the problem.
I will take some action and make an announcement by next Monday.

ytanimura, 2021-02-19:
Now I fixed the bug in my environment and push to master as bfbe25dfde69cca31ae569842d0e97fafa75eb8b .

If DX12 is selected as the backend, all variables output from the vertex shader must be read by the fragment shader.

I added a line `layout(location = 1) in vec2 _uv;` to  `polygon.frag` and tests with specifying backends.

If you also confirm the bug fix, this issue will be closed.

daidanretsu, 2021-02-20:
Thank you for quick fixing the issue I reported. I will check it and get back to you. But the confirmation may be next Wednesday.

daidanretsu, 2021-02-23:
I checked and the example worked fine in my environment! Thank you,

ytanimura, 2021-02-24:
Thank you for your kindly report!
Now, we close this issue.

closed 2021-02-24 by ytanimura