ricosjp/truck #11: Convert surface with boundaries to triangle mesh

J-F-Liu, 2021-05-13:
Is this implemented or not?
I only find the Surface defination:
```rust
pub enum Surface {
    /// Plane
    Plane(Plane),
    /// 3-dimensional B-spline surface
    BSplineSurface(BSplineSurface<Vector3>),
    /// 3-dimensional NURBS Surface
    NURBSSurface(NURBSSurface<Vector4>),
    /// revoluted curve
    RevolutedCurve(Processor<RevolutedCurve<Curve>, Matrix4>),
}
```

ytanimura, 2021-05-19:
We have a plan, and we are experimenting with it whenever we have free time. However, I can't promise how long it will take, because we give higher priority to the implementation of STEP I/O, Euler operations, and set operations with the boundary representation.

J-F-Liu, 2021-05-19:
I already know B-spline and NURBS surface, can you recommand some readings about mathmatical theory of B-Rep surfaces? Thanks.

ytanimura, 2021-05-19:
Sorry, but I can't give you the best answer to this question. In fact, I have yet to find a definitive book on the subject. I tend to extrapolate from the specifications of existing CAD kernels, including OCCT. As for set operations, I plan to use [this book](https://www.springer.com/gp/book/9781846283123) as a reference, but I haven't compared it to several other books either.

ytanimura, 2021-05-24:
Now, [a prototype version](https://github.com/ricosjp/truck/blob/master/truck-meshalgo/src/tessellation/triangulation.rs) has been implemented.
The difficult part of the Delaunay triangulation is left to the crate [spade](https://crates.io/crates/spade). This implementation is CPU-driven and may not provide the performance needed as the model grows in size.
Currently, we are implementing some additional mesh algorithms for the test of this tessellation.

J-F-Liu, 2021-07-02:
I suspect that the CDT algorithm in spade cannot handle holes inside a polygon.
The book _Boundary Representation Modelling Techniques_ lacks detailed algorithms.
I found the following paper which gives an appropriate solution to the problem:

[Advancing Front Surface Mesh Generation in Parametric Space Using a Riemannian Surface Definition](http://ansys.net/ansys/papers/meshing/advanced_front_tri.pdf)

ytanimura, 2021-07-04:
> I suspect that the CDT algorithm in spade cannot handle holes inside a polygon.

That is right, If one specifies a boundary, the triangle will always be entirely outside or inside the domain. So, the current implementation excludes triangles whose center of gravity is outside the domain when converting data from spade to truck mesh structure. Although there is overlap, we consider that speeding up the process is an issue to be addressed after we have some functionality in place.

> I found the following paper which gives an appropriate solution to the problem:

Thank you for the introduction. I will check it out in the near future.

J-F-Liu, 2021-07-23:
Found another paper: [Gabriel-constrained Parametric Surface Triangulation](https://www.semanticscholar.org/paper/Gabriel-constrained-Parametric-Surface-Ruiz-Cadavid/bc96519380a481b1f43ff64d28f3048846b411e3)

This approach looks better, only the algorithm to generate sample points is very slow, needs improvement.

g-rauhoeft, 2021-07-28:
There is also this rust crate that looks very promising, which implements Manifold Dual Contouring: https://github.com/hmeyer/tessellation