ricosjp/truck #13: T-Splines

giannissc, 2021-07-30:
Are there plans to support t-splines as well on truck?

Also seeing how t-splines are a superset for NURBS and subdivision surfaces (hope my understanding is correct :p ) would it make sense to make all modelling centre around T-splines?

ytanimura, 2021-07-30:
T-spline is patented by Autodesk. https://patents.google.com/patent/US7274364B2/en
It seems to use it with Rhino. We need to check the conditions of availability. It is not immediately supported.

johschmitz, 2023-03-08:
I agree with @virtualritz. Could the term "subdivision surface" or "subdiv" be added to the issue title in order to avoid duplicates?

ytanimura, 2023-03-23:
One motivation for creating a shape processor from scratch is to provide a wasm module that can be handled in the browser for small models. There are cases of transient code that cannot be made wasm, or cases where the code is parallelized and accelerated only in native, but basically everything is ideally designed to work in the browser. Maybe I am just uninformed, but I think it is reasonably hard to run code that includes FFI on a browser. Therefore, even if it were to be implemented, it could only be used in a very limited way, and I am not very positive about it.

virtualritz, 2023-03-23:
Yeah, no idea how hard it would be to [cross compile C++ to wasm](https://developer.mozilla.org/en-US/docs/WebAssembly/C_to_wasm) that later sits behind an FFI for Rust.

That said, I don't think it's an issue, at the beginning, if a feature like subdivision surface support sits behind a feature flag that must be disabled to build `truck` for wasm.

lynaghk, 2024-04-05:
The linked T-spline patent is now expired.

eliasboegel, 2024-06-27:
T-splines would be extremely useful for me. If thats a feature that could be developed, that would be phenomenal!

virtualritz, 2024-06-27:
See [my comment above](https://github.com/ricosjp/truck/issues/13#issuecomment-1436900967). I think it much more useful to have support Catmull-Clark-subdivision surfaces-as-patch meshes.

What would T-Splines enable that CC-subdivision surfaces would not?

eliasboegel, 2024-06-27:
> See [my comment above](https://github.com/ricosjp/truck/issues/13#issuecomment-1436900967). I think it much more useful to have support Catmull-Clark-subdivision surfaces-as-patch meshes.
> 
> What would T-Splines enable that CC-subdivision surfaces would not?

Please correct me if I'm wrong here. My understanding is that CC subdivision surfaces can't represent exact conical sections because these surfaces are still described by non-rational splines. Given that NURBS are a special case of T-splines, they can handle this use-case.

virtualritz, 2024-06-27:
> My understanding is that CC subdivision surfaces can't represent exact conical sections because these surfaces are still described by non-rational splines.

That is true. However, what are the practical implications? If you design a part that has conical sections that need to be perfect (vs approximated to something not distinguishable for the human eye) you are likely solving an engineering problem, not a design problem.

I.e. you are working on a mechanical part and you would  simply use actual NURBS geometry for these parts.

I guess I'm saying: show me a part where using T-splines made more sense than using a combination of NURBS and a connected patch meshes originating from a CC-sds. :)

Also note that T-splines have topological limitations (quads only) that CC-sds do not (and which specifically matter when you design organic stuff, e.g. product design).

The people at the forefront of T-spline reasearch have specifically have many and also their [latest paper](https://arxiv.org/pdf/2103.05726) scattered with examples that are exactly suited to be done with existing sds-technology. I.e. they don't require T-splines ...

virtualritz, 2024-06-28:
Fair enough. I didn't know and I see your point. 
I also learned something. Thank you for taking the time to explain your motivation.

I will open a separate issue for sds support in `truck`. ☺️

PeterDessev, 2024-07-08:
Hey all, though I would add that I am currently working on T-mesh and T-NURCC implementations for `truck`. The fork is public and can be seen [here](https://github.com/PeterDessev/truck). Let me know what you guys think. 

Currently, I have only implemented some of the very rudimentary logic for T-meshes as described in the original paper on the subject, however, more is on the way!

virtualritz, 2024-07-08:
This is awesome
I do have to bring the naming issue upmagain though.

Unlike pretty much any other language, Rust has [clear guidelines](https://rust-lang.github.io/api-guidelines/naming.html#casing-conforms-to-rfc-430-c-case) on naming structs.

E.g. it should be `Tspline`, not `TSpline`.

See also this [post](https://users.rust-lang.org/t/naming-convention-suggests-name-that-is-read-incorrectly/55002/4) on the rust mailing list which IMHO nails why this is actually great and shouldn't be a point on contention, even if you're not a fan (I wasn't at the beginning but have hence changed my mind). 😁

saosebastiao, 2025-01-03:
Just want to add the comment that the T-Spline patent is now fully expired:
https://patents.google.com/patent/US7274364B2/en