ricosjp/truck #14: Cylinder showing problem

happydpc, 2021-08-14:
I test the cylinder code following:
```rust
pub fn create_cylinder(height: f64, radius: f64) -> Solid {
    // make a solid cylinder
    let vertex = builder::vertex(Point3::new(0.0, 0.0, radius));
    let circle = builder::rsweep(&vertex, Point3::origin(), Vector3::unit_y(), Rad(7.0));
    let disk = builder::try_attach_plane(&vec![circle]).unwrap();
    let solid = builder::tsweep(&disk, Vector3::new(0.0, height, 0.0));
    // Return the solid as a boundary shell for easier processing later.
    // solid.into_boundaries().pop().unwrap()
    solid
}
```

With default viewer, this comes out this. And some corner shows shadow lines with some camera angle.
[image]

ytanimura, 2021-08-16:
Thank you very much for your information. I was able to reproduce the problem in my environment. It seems to be a bug when trimming the mesh. We will fix it before the release of v0.3.

happydpc, 2021-10-11:
I found the problem may be the depth issue. Using the wgpu reverse z-depth method can solve this problem.

ytanimura, 2021-12-04:
The bug here seems to have already been fixed. I went back through the history and found that the following two commits seemed to have improved the symptoms.

https://github.com/ricosjp/truck/commit/444fdec2b2ff522cd586255b3764ff5dd9a2ce55
https://github.com/ricosjp/truck/commit/65be75cf2532b23f59681483b668d78954d4eea8

As for the bug, it was still a mesh bug. A large triangle overlapping a plane was generated. The same problem occurred with the standard Windows 3D viewer when I read the mesh generated from the commit before the ones above. The reverse depth method probably prevented large triangles from being drawn.

[image]

ytanimura, 2021-12-04:
Now, v0.3 was released. This issue should be closed.

closed 2021-12-04 by ytanimura