ricosjp/truck #15: Large number in creating shapes

happydpc, 2021-08-17:
When using rsweep with large number for vertex coord, for example, 500 for x. It will panic:

```rust
panicked at 'failed to create instance', truck-rendimpl\src\shaperend.rs:86:14
```

```rust 

fn create_test_solid() -> Solid(){

 let vertex = builder::vertex(Point3::new(500.0, 0.0, 100.0 as f64));
        // sweep the vertex along a circle
        let circle: Wire = builder::rsweep(
            // the reference to the vertex
            &vertex,
            // a point on the axis
            // Point3::new(d.offset as f64 + d.radius as f64, 0.0, 0.0),
            Point3::new(500.0 as f64, 0.0, 0.0),
            // the direction of the axis
            Vector3::unit_y(),
            // If the absolute value is no less than 2π radian, a closed shape will be generated.
            Rad(7.0),
        );
        let disk = builder::try_attach_plane(&vec![circle]).unwrap();
        // sweep the circle along a circle
        let mut solid = builder::rsweep(
            // the reference to the wire
            &disk,
            // a point on the axis
            Point3::origin(),
            // the direction of the axis
            Vector3::unit_z(),
            // If the absolute value is no less than 2π radian, a closed shape will be generated.
            Rad(d.angle as f64),
        );
        solid
}
```

ytanimura, 2021-08-23:
Thank you very much for your report.

We investigated the cause of the problem, and it seems that the Newton method is not converging with the required accuracy when finding the texture coordinates of the boundary curve.

We already know the ad-hoc solution, but I am still trying to decide the best choice for the overall design. We will try to come to a conclusion in a little while.

ytanimura, 2021-10-03:
The following two actions have been taken.
- Convergence judgment of Newton's method has been changed to one using derivatives (the judgment condition has been weakened to a feasible accuracy).
- The meshing accuracy of simple-shape-viewer is now set by the relative error from the bounding box.

We have saved the results of modeling with the given function in [large-torus.json](https://github.com/ricosjp/truck/blob/master/truck-meshalgo/tests/tessellation/large-torus.json). One can test this issue by launching simple-shape-viewer and dragging and dropping the large-torus.json file. Since it is essentially a tessellation bug, CI only tests the tessellation part.

ytanimura, 2021-12-04:
Now, v0.3 is released. This issue should be closed.

closed 2021-12-04 by ytanimura