ricosjp/truck #17: Boolean

rustythirsty, 2021-08-24:
Any hint as to when we can expect Boolean operations to be made available in the package.  I would like to know if it is possible to generate a mesh from a only a set of 3D points.

ytanimura, 2021-08-25:
We will be announcing more details about the progress of the Boolean operations in this week. There are some performance issues, but so far the implementation is making steady progress.
There are no concrete plans for generating a mesh from point cloud at this time. We plan to develop the API for web applications once STEP output and set operations are in place.

ytanimura, 2021-08-30:
Current Status:
Now we are implementing Boolean operations for solids in the case where all interfering faces are transversely intersected. We already implemented:
- extraction of intersection polyline curves between two polygon meshes,
- the struct `IntersectionCurve` for describing intersection curve between two surfaces, (Each intersection curve is leaded by a polyline curve extracted the above function.) and
- creation of the loop wires (topological element) by collecting existing edges and new edges which consist the above intersection curves.

The picture below shows the polygons of the common part of the two parabolas, achieved by ad-hoc choosing the extracted loop wires.
[image]

Future tasks:
- selection of the loop wires included in an AND , OR or MINUS shape of two solids
- stabilization and acceleration of implementation. Now, it is unstable so that the unit tests fail 30% of the time and the constructed polygon have some holes
- Implementation for the case where some faces are in close contact with each other

ytanimura, 2021-10-02:
The basic implementation has been completed, and the operation has become reasonably stable, although not yet at the enterprise level. The ruststep has been released, so the set operations will be documented, and work on STEP I/O will be resumed first.

MattFerraro, 2024-02-13:
You mention:
> Implementation for the case where some faces are in close contact with each other

My current project, a CAD program, is unable to proceed until this feature is implemented. Is there any way I can help to get it done?

qff233, 2024-06-14:
How to boolean operate for mesh? truck-meshalgo has not this operation. TAT