ricosjp/truck #19: How to draw "helper" 3d lines ?

rsaccon, 2021-11-16:
What is the best way to draw with truck-platform simple 3d lines (or other simple objects) for things like visualizing the coordinate system (e.g. lines with arrows along the coordinate axes and a grid at the xz-plane). And as requirement for lines I would like to set their color and thickness if possible. truck does not seem to expose an API for specific for these tasks but with` truck-modeling::builder` I think I can construct anything. Or is there any other more appropriate approach ?

ytanimura, 2021-11-16:
This may be a less than desirable answer, the most natural solution to coordinate system visualization is to create a new drawing element using the low-level API of wgpu. There is no current render objects in `truck` that would be fixed to the camera movement; by implementing [`Rendered`](https://github.com/ricosjp/truck/blob/master/truck-platform/src/lib.rs#L304) trait using the wgpu's low-level API, one can build our own assets. [`wgsl-sandbox`](https://github.com/ricosjp/truck/blob/master/truck-platform/examples/wgsl-sandbox.rs) in truck-platform is such an example.
We are looking into creating helpers for things like coordinate systems in the future. However, to be honest, we don't have a lot of resources to devote to visualization, so we can't say exactly when that will happen. The reasons for not being able to devote resources to the visualization side include the high priority of the shape processing algorithm and the fact that wgpu is unstable and cannot be incorporated into concrete services for the time being.

rsaccon, 2021-11-16:
Thanks a lot for the insights. I will take a look at `wgsl-sandbox`. And I look forward to the progress on the shape processing algorithm, which is the main reason I am tinkering around with truck. So I am gonna close this.

closed 2021-11-16 by rsaccon