ricosjp/truck #20: Trouble with builder::try_attach_plane

rsaccon, 2021-11-22:
at me experiments with truck I ran into an issue where `builder::try_attach_plane` sometimes worked as expected and sometimes just failed with `WireNotInOnePlane`. I extracted from my app some actual numerical values where it fails

FAIL:
``` Rust
        let start = Point3::new(333.3333333333333, 0.0, 0.0);
        let axis = Vector3::new(333.3333333333333, 0.0, 0.0);
        let pt = Point3::new(333.3333333333333, 0.0, 25.0);
        let vertex = builder::vertex(pt);
        let circle = builder::rsweep(&vertex, start, axis, Rad(7.0));
```
I "rounded" the numbers a bit and then it worked: 
OK
``` Rust
        let start = Point3::new(333.3333333333333 as f32 as f64, 0.0, 0.0);
        let axis = Vector3::new(333.3333333333333 as f32 as f64, 0.0, 0.0);
        let pt = Point3::new(333.3333333333333 as f32 as f64, 0.0, 25.0);
        let vertex = builder::vertex(pt);
        let circle = builder::rsweep(&vertex, start, axis, Rad(7.0));
```
But like this I lose precision and I a have no clue how reliable it is

So how can I make `builder::try_attach_plane` work in a reliable way ?

ytanimura, 2021-11-25:
I'm sorry it's not specified in the documentation, but the rotation axis must be normalized; I was able to run it by changing `axis` to `Vector::unit_x()`.

rsaccon, 2021-11-25:
OK, thanks, gonna close this then.

commit 6e306ac6 (2021-02-16): Merge branch '20-nurbssurface-more-methods' into 'master'

more methods for nurbs

Closes #20

See merge request ricos/truck/truck!89
closed 2021-11-25 by rsaccon