ricosjp/truck #23: How to handle multiple PolygonInstances

rsaccon, 2021-12-09:
I need some clarification on best approach for creating a scene with repeating objects which share the same base shape (e.g. a cylinder). Let's say I have about hundred of these cylinders and their form, position and orientation is defined by their model matrix of their PolygonInstance or WireframeInstance. So if I create just one PolygonInstance and WireframeInstance and then clone it hundred times, and of course adapt their model matrix, does that mean they still share all the same Mesh Data  ? Obviously, for performance and memory saving reasons that is what I want. But I am not sure, if I understood this properly.

ytanimura, 2021-12-10:
`PolygonInstance` can be duplicated by the [`clone_instance`](https://docs.rs/truck-rendimpl/latest/truck_rendimpl/struct.PolygonInstance.html#method.clone_instance) method, and it will be duplicated as an instance that shares the mesh data on the GPU. Each cloned instance can have its own material and matrix.
Internally, all the shapes are drawn by looping through the number of instances that have been created. This means that instancing with multiple slots is not supported at this time. Memory optimization has been done at a minimum, but the number of instructions to GPU has not yet been optimized.

rsaccon, 2021-12-10:
Ok, thanks a lot for the detailed explanation.

closed 2021-12-10 by rsaccon