ricosjp/truck #24: Can we make truck_platform::Camera.projection public

rsaccon, 2021-12-09:
The reason is the following: I tried to use this [3d gizmo](https://github.com/urholaukkarinen/egui-gizmo), it works perfectly (beside of some flickering caused by my egui integration), but that gizmo needs the projection matrix. I could calculate that matrix from `Camera.projection()` and `Camera.matrix` or even mess around with egui-gizmo sources, but the easiest is just to have access to the raw projection matrix.

ytanimura, 2021-12-13:
Sorry for the late reply. As it turns out, it does not meet your expectations.
There is already a method with the same name, which is more suitable for the purpose. The internal variable "projection" is not a matrix with a name that can be neatly explained. If I’m forced to say, it is a projection matrix for a fixed camera in the origin and a 1:1 screen aspect ratio. In the future, we may save the fov and clipping distance and discontinue this one. If you really need it, we think the natural way to use the API is to set the aspect ratio to 1 and then multiply the camera's matrix.

rsaccon, 2021-12-13:
Ok, that's fine.

closed 2021-12-13 by rsaccon