ricosjp/truck #25: How to update WireFrameInstance ?

rsaccon, 2021-12-14:
for updating the mesh at a `PolygonInstance`  I do:

`my_polygon_instance.swap_vertex(&mut another_polygon_instance);` 

for updating the vertices and indexes of a `WireFrameInstance` there is no such equivalent.  I tried to do it myself with something like:
```
my_wire_instance.vertices = another_wire_instance.vertices.clone();
my_wire_instance.strips = another_wire_instance.strips.clone();
```
but that is not possible, because the `vertices` and `strips` are private.

Am I missing something ?

ytanimura, 2021-12-14:
Thank you for your issue. Certainly, there is not such a method. We will add it. Please wait a little.

ytanimura, 2021-12-15:
Now added the method. https://github.com/ricosjp/truck/blob/master/truck-rendimpl/src/wireframe_instance.rs#L33

rsaccon, 2021-12-15:
Great, thanks.

closed 2021-12-15 by rsaccon