ricosjp/truck #33: How to export sphere to stp file

happydpc, 2022-09-04:
According to the code, there are two ways to create a sphere:
```rust
let uhcircle = NURBSCurve::new(BSplineCurve::new(knot_vec, control_points));
    let sphere = RevolutedCurve::by_revolution(
        uhcircle, Point3::origin(), Vector3::unit_x(),
    );
```
```rust
let sphere = Sphere::new(center, radius);
```

but they all don't impl the ModelShape trait, how to export sphere to step file ?

happydpc, 2022-09-21:
Good job on this work, Thanks.

ytanimura, 2022-09-23:
I am very sorry for the delay. One can output the sphere step by the following code and it displays properly in FreeCAD for example.

```rust
use std::f64::consts::PI;
use truck_modeling::*;
use truck_stepio::out;

fn main() {
    let vertex = builder::vertex(Point3::new(0.0, 0.0, 1.0));
    let wire = builder::rsweep(&vertex, Point3::origin(), Vector3::unit_y(), Rad(PI));
    let shell = builder::rsweep(&wire, Point3::origin(), Vector3::unit_z(), Rad(PI * 2.0));
    let step_string =
        out::CompleteStepDisplay::new(out::StepModel::new(&shell.compress()), Default::default())
            .to_string();
    std::fs::write("output.stp", &step_string).unwrap();
}
```

However, in truck it is a bit abuse and does not close spherical as topological data. We would like to fix this, but we need to take care of [a problem](https://github.com/ricosjp/truck/issues/5) at the same time before we can officially support it.

joeblew999, 2026-04-05:
I'm a downstream user building on truck — sharing a reference from the [monstertruck](https://github.com/virtualritz/monstertruck) fork.

The fork has full STEP export implemented for `Sphere` in [`monstertruck-step/src/save/geometry.rs`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-step/src/save/geometry.rs#L481):

```rust
impl DisplayByStep for Sphere { ... }                     // L510
impl StepSurface for Sphere {}                            // L516
impl DisplayByStep for Processor<Sphere, Matrix4> { ... } // L481 — handles transforms
```

The `Processor<Sphere, Matrix4>` impl correctly handles non-uniform scale detection, orientation, and outputs `SPHERICAL_SURFACE` with a proper `AXIS2_PLACEMENT_3D`. STEP import for `Sphere` is also implemented in [`src/load/step_geometry/sphere.rs`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-step/src/load/step_geometry/sphere.rs) with `ParametricSurface`, `SearchNearestParameter`, and `ParameterDivision2D` implementations.

Apache-2.0 — clean reference for upstream STEP sphere support.