ricosjp/truck #36: Speeding up triangulation

alxpettit, 2022-12-06:
Currently the `MeshableShape` implementation in Truck is quite slow if you set `tol` high, but the whole time, it's only using one CPU core. Given the propenderence of iterators and maps in the function, and in the `tesselation` one it calls, would something like [rayon](https://crates.io/crates/rayon) be a good solution to parallelizing the operation?

ytanimura, 2022-12-07:
Actually we are thinking the same thing and have completed the technical verification. We have not yet been able to merge it into the master branch because of the following minor issues that remain.

- The struct [`ID`](https://docs.rs/truck-base/latest/truck_base/id/struct.ID.html), which widely used by topological algorithms, is not implemented Send and Sync because the raw pointer is used as the actual state.
- One of the goals of truck is to run in wasm. We recognize that parallel processing on a browser has a large overlap; in the case of wasm, it is necessary to apply a branching process that does not parallelize the process.

In fact, it won't take much time. Now that we have received the issue, we will try to merge it into master within the next week or a little later.

ytanimura, 2022-12-13:
Now, `MeshableShape::triangulation` has been parallelized!

There is a [benchmark](https://github.com/ricosjp/truck/blob/master/truck-meshalgo/src/tessellation/triangulation.rs#L354) that compare new algorithm to old one.

closed 2023-01-27 by alxpettit