ricosjp/truck #37: Support for 2D sketching/constraint solving

MattFerraro, 2023-01-03:
Hello,

I'm very excited about this project! Do you plan to implement your own 2D sketching and constraint solving? This is on its own quite a difficult thing to get right, judging by the [review](https://www.researchgate.net/publication/232654491_2D_Geometric_Constraint_Solving_An_Overview) [papers](https://arxiv.org/pdf/2202.13795.pdf) I've seen.

One possibility could be to use [Solvespace](https://solvespace.com/index.pl) as a [library](https://github.com/solvespace/solvespace), but it's written in C not Rust.

I'd be interested in collaborating with you on this, if you decide to write your own!

ytanimura, 2023-01-05:
We have actually created a tool using truck that draws a simple region in 2D and extrudes it out as a STEP. ([Introducing tweet](https://twitter.com/RICOS_ltd/status/1550390552482693120?s=20&t=zJGRTi1ajl_ik7HiacZmPg) (Japanese)) However, the source code for this one is not open to the public.
Solvers at intersections are indeed difficult and have not yet been created. We have to admit that it is a lower priority than STEP for the time being. I will take a look at the materials you provided sometime.

happydpc, 2023-12-15:
> On the note of 2d drawing / constraint solving, I've been working on my own in Rust / egui and came across this issue! I'm hoping to use truck to generate STL's, if I can figure out how to get the library to consume an outer and many inner polygons and tessellate + export to STL.
> 
>  recording.mp4 
> This is what I've made so far, I hope to open source soon: [liquid-cad.twitchyliquid.net](https://liquid-cad.twitchyliquid.net/)

That's cool, can't wait to have a try.

emfax, 2024-09-19:
If people are interested in working on a 2D constraint solver, I would be keen to help. I’ve also read some of the recent papers and it does seem difficult, but doable. Maybe some of the authors of those papers could provide some guidance.

Something that might need to be looked at in parallel would be some sort of lightweight computer algebra system that could support basic mathematical expressions.

I’ve been looking at this for a while, would be great to work on something with others.

twitchyliquid64, 2024-09-21:
@emfax If you want to collab I would recommend talking with @MattFerraro who is actively working on https://github.com/CADmium-Co/CADmium.

As for my foray into this space, i made and have now put into maintenance mode https://github.com/twitchyliquid64/liquid-cad

 - For the solver: https://github.com/twitchyliquid64/liquid-cad/blob/main/eq/src/dumbass_solve.rs#L172
 - For the symbolic algebra system: https://github.com/twitchyliquid64/liquid-cad/blob/main/eq/src/lib.rs#L151

emfax, 2024-09-22:
@MattFerraro I’d interested in working on a 2D constraint solver. I’ve been through a bit of the recently published literature as well and I’ve contacted some authors. I think it could be worthwhile trying to get something rudimentary working at the very least.