ricosjp/truck #38: WASM/Javascript bindings

MattFerraro, 2023-01-03:
Are you planning to create WASM builds and javascript bindings for truck? I think it would be very useful as a way of showcasing what the library is capable of. I'd be very interested in using such bindings if they existed to build an open source web-based CAD program!

ytanimura, 2023-01-05:
Thank you for your issue.

[truck-js](https://github.com/ricosjp/truck/tree/master/truck-js) is a js-binding with wasm. [Simple modeling](https://github.com/ricosjp/truck/blob/master/truck-js/tests/test.js) can be done. It is difficult to wrap all low-function APIs because generics are not available, but we would like to have a STEP-related high-function API. We would like to bind `Scene` and such, but we are holding off due to lack of testing because WebGPU is not quite stable.

MattFerraro, 2024-02-13:
Thank you, that's perfect!

closed 2024-02-13 by MattFerraro