ricosjp/truck #39: Canonical Struct naming

virtualritz, 2023-02-20:
See [Rust API naming guidelines](https://rust-lang.github.io/api-guidelines/naming.html).

> In `UpperCamelCase`, acronyms and contractions of compound words count as one word: use `Uuid` rather than `UUID`, `Usize` rather than `USize` or `Stdin` rather than `StdIn`.

For example in `stl.rs`:

`STLFace` -> `StlFace`
`STLReader` -> `StlReader`

`STLType` -> `StlType`
`STLType::ASCII` -> `StlType::Ascii`

etc.

Would you consider a PR with such changes if I opened one?

ytanimura, 2023-02-21:
Thank you for your issue.

If there is something official as evidence, we will try to follow it. It would be helpful if you could give me a pull request.

We would like to specify in advance. For single letter + complete word combinations, we will consider the first letter and the complete word as separate words. Official Rust APIs include, for example, `BTreeMap`. For example, in `truck`,

- Use `BSplineHoge` rather than `BsplineHoge`.
- Use `PCurve` rather than `Pcurve`.

virtualritz, 2023-02-23:
> We would like to specify in advance. For single letter + complete word combinations, we will consider the first letter and the complete word as separate words.

Yes, I complete agree with this. I would only rename structs like the one I mentioned.

virtualritz, 2023-02-23:
Another thing I noted is the naming of composite `structs`. I.e. usually the type of geometry the `struct` holds is at the end of the name. For example: `NurbsSurface`. It holds a surface.

But `RevolutedCurve` also holds a surface. So it should be named `RevolutedSurface` or `RevolutedCurveSurface`?
Or am I missing something?

ytanimura, 2023-02-24:
Well, that's what I was struggling with too. What is being rotated here is a curve, not a surface, and a `RevolutedSurface` should point to S^2xS^1 in R^4, for example. However, `SurfaceOfRevolution` is a little less than pleasing to the ear. I wish there was some more natural way to describe it.

virtualritz, 2023-03-06:
On that note. I saw `is_geometric_consistent()`. I think this should be `is_geometrically_consistent` or maybe it could be abbreviated to `is_geo_consistent()`?

I.e. there is the geometric consistency and if that is true then the resp. shape is geometrically consistent.