ricosjp/truck #42: How to convert Solid or Shell to triangle mesh for rendering?

MattFerraro, 2023-02-27:
I've been able to generate Solids and Shells in Rust, but I'd like to create a triangle mesh that I can send to three.js for rendering. Clearly truck is capable of generating that mesh, since we can view the bottle with webGPU, but I can't figure out how to do it! Could you offer some pointers?

ytanimura, 2023-02-27:
Thank you for your question.

There are [APIs](https://github.com/ricosjp/truck/blob/master/truck-js/src/polygon.rs) for handling meshes with JavaScript.

There is an [example](https://github.com/ricosjp/truck/tree/master/truck-js/examples) for using truck with WebGL. In [`script.js`](https://github.com/ricosjp/truck/blob/master/truck-js/examples/script.js),
- modeling  cube with truck API, meshing and getting Vertex-Index buffers (L23 - L32),
- rendering with the above buffer and shaders in `index.html` (function `onLoad` and `render`).
- reading json or STEP (function `fileRead`), and
- downloading obj (function `downloadObj`).

MattFerraro, 2023-03-01:
incredible, thank you!

commit 07079e81 (2020-12-27): Merge branch '42-cmake-readme' into 'master'

write "depends on cmake" to README

Closes #42

See merge request ricos/truck/truck!54
closed 2023-03-01 by MattFerraro