ricosjp/truck #45: `README.md` in each crate

ytanimura, 2023-03-03:
Consider for `README.md` in each crate cf. [README.md by virtualritz](https://github.com/virtualritz/truck/tree/master/truck-rendimpl)

MattFerraro, 2023-03-05:
I would love to see a readme that describes in a few sentences the primary data types and when you're supposed to use which ones. There seems to be a few like `Vector3`, `Point3`, `Vertex`, which all seem similar. There's also things like `Edge` and `Wire` and `Line`, and I'm not sure when I should use one over another.

ytanimura, 2023-03-06:
Thank you for your comment.

Would it be sufficient to write something similar to the beginning of [the `truck-topology` document] in each crate README?

[the `truck-topology` document]: https://docs.rs/truck-topology/latest/truck_topology/

bigmark222, 2025-11-15:
check out my pr

joeblew999, 2026-04-05:
I'm a downstream user building on truck — sharing a reference from the [monstertruck](https://github.com/virtualritz/monstertruck) fork (the same virtualritz referenced in the issue).

The fork has now done this comprehensively — every crate has a README with a description, quick-start example, and feature table:

- [`monstertruck-geometry`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-geometry/README.md) — knot vectors, B-splines, NURBS, T-splines
- [`monstertruck-solid`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-solid/README.md) — boolean ops, fillets, shape healing, with working quick-start code
- [`monstertruck-topology`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-topology/README.md) — vertices, edges, wires, faces, shells, solids
- [`monstertruck-step`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-step/README.md) — STEP import/export
- [`monstertruck-wasm`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-wasm/README.md) — WASM/JS bindings
- [and all remaining crates](https://github.com/virtualritz/monstertruck/tree/682c67622b4a04863fc4151c63ddcd07c6b4bb69)

All Apache-2.0. The format and style might be useful as a template for upstream truck crate READMEs.