ricosjp/truck #46: Replace `Mutex` with `parking_lot::Mutex` and `Arc` with `rclite::Arc`

virtualritz, 2023-03-06:
Rationale:

* [`parking_lot::Mutex`](https://docs.rs/parking_lot/latest/parking_lot/type.Mutex.html) is the unwritten actual standard for various reasons (see the link). It also makes the code not being littered with `unwrap()`s wherever a `Mutex` is locked. I.e. you will find that almost all crates in the wild that need `Mutex` or `RwLock` will use the ones from `parking_lot` and not the ones from `std`. Using this is kind of a one way street as adding a feature flag means a lot of code duplication and lots of clutter in the code (i.e. worse than the aforementioned `unwrap()`s.

* [`rclite::Arc`](https://docs.rs/rclite/latest/rclite/) is purportedly faster than the `std` variant and it also uses a lot less memory (4 bytes on 64bit systems vs 16 bytes for the `std` variant). This could be hidden behind a feature flag as the access syntax is the same.

virtualritz, 2023-03-06:
If you agree I am happy to do the required changed and open a PR.

ytanimura, 2023-03-07:
Thank you for the introduction. We would love to include it. These crates have a lot of downloads and seem reliable. There are a few things we are concerned about, such as behavior with wasm, but those can be found by turning CI in our GitLab. Please send me a pull request.

virtualritz, 2023-03-07:
> Please send me a pull request.

Done! It's now part of #48.

virtualritz, 2023-03-08:
Fixed with merge of #48.

commit 8585694f (2021-02-16): Merge branch '46-wire-frame-rendering' into 'master'

Resolve "エッジの描画"

Closes #46

See merge request ricos/truck/truck!91
closed 2023-03-08 by virtualritz