ricosjp/truck #53: Support for Chamfer/Fillet

MattFerraro, 2023-12-21:
Hello,

I'd love to see support for Chamfer and Fillet commands within Truck. Is that on the roadmap? What do you think it would take to implement?

Cheers

ytanimura, 2024-01-05:
Thank you for your question.

The fillet creation will be implemented at some point, but we are not yet in a position to factor it into our concrete plans.

twitchyliquid64, 2024-10-24:
I wonder if its possible to do this today with the `Shell.cut_edge()` method?

1. Identify every edge which forms a wire for the corners you want to chamfer
2. For each edge, use `cut_edge` twice (once for topside, once for bottom side) to make a new edge, offset from the corner by the chamfer distance
3. Update the previous edges to be colinear with the sloped/chamfered surface, basically redundant
4. .. vibes?

commit 81e58f55 (2021-01-27): Merge branch '53-z-plane-attach' into 'master'

Resolve "法線がz軸正向きのときのplane attachがうまくいっていない"

Closes #53

See merge request ricos/truck/truck!69