ricosjp/truck #55: STL / obj write produces files with errors

twitchyliquid64, 2024-01-10:
Consider a 2d triangle/polygon with a smaller 2d triangle cut out of it:

[image]

When I extrude this using `tsweep` and then generate a STL / OBJ file with it and import it into PrusaSlicer, the slicer reports errors:

[image]
[image]

I've tried compressing the shape, running `put_together_same_attrs` on it, and a number of variations for how to create the wires/faces, but all of them seem to create files that report errors.

Am I doing something wrong here? Is it possible there is a bug in truck?

### Appendix: Rough code for generating above

I guessed the co-ordinates, but this is the general gist

```rust
let points = [
  // outer triangle
  builder::vertex(Point3::new(0.0, 0.0, 0.0)),
  builder::vertex(Point3::new(5.0, -5.0, 0.0)),
  builder::vertex(Point3::new(10.0, 0.0, 0.0)),  
  // inner triangle
  builder::vertex(Point3::new(1.0, 1.0, 0.0)),
  builder::vertex(Point3::new(5.0, -2.0, 0.0)),
  builder::vertex(Point3::new(9.0, 1.0, 0.0)),
];

let outer_wire: Wire = vec![
  builder::line(&points[0], &points[1]),
  builder::line(&points[1], &points[2]),
  builder::line(&points[2], &points[0]),
].into();
let inner_wire: Wire = vec![
  builder::line(&points[3], &points[4]),
  builder::line(&points[4], &points[5]),
  builder::line(&points[5], &points[3]),
].into();

let face = builder::try_attach_plane(&vec![outer_wire, inner_wire]).unwrap(); // I've also tried face.add_boundary(inner_wire)
let solid = builder::tsweep(&face, height * Vector3::unit_z());

use truck_meshalgo::tessellation::MeshableShape;
use truck_meshalgo::tessellation::MeshedShape;
let mesh = s.triangulation(0.05).to_polygon();

// use truck_meshalgo::filters::OptimizingFilter;
// mesh.put_together_same_attrs();

let mut out = Vec::with_capacity(1024);
truck_polymesh::stl::write(&mesh, &mut out, truck_polymesh::stl::STLType::Binary).unwrap();
```

twitchyliquid64, 2024-01-22:
I was poking around and found the `remove_degenerate_faces` and `remove_unused_attrs` functions, I tried them like:

```rust
use truck_meshalgo::filters::OptimizingFilter;
mesh.put_together_same_attrs().remove_degenerate_faces()
    .remove_unused_attrs();
```

and it seems to work?

If so, could we document all the steps necessary to tessellate a truck solid to something like an STL?

twitchyliquid64, 2024-01-24:
One more issue I've discovered: If your wires form counter-clockwise loops, the STL breaks. Fortunately that is easy to fix by detecting if the signed area is negative, and calling `.invert()` on the wire before use.

ytanimura, 2024-01-26:
I am very sorry for the delay in replying.

First, looking at the code you gave me, the outer triangle is not in the inner triangle. This could be a simple mistake, since this is meant to be a rough code.

As a more essential matter, the orientation of the boundary must be strictly specified. The outer wires should be specified counterclockwise and the inner wires clockwise. The order in which the wires are specified is of no concern at all. The trimming of surfaces is done to include the left side of the wire's direction of travel. `truck` is designed to be an easy-to-engineer base library and requires the implementation of "user-friendly" modifications to wire orientation to be done by the product being built.

# Corrected Code
```rust
use truck_meshalgo::prelude::*;
use truck_modeling::*;

fn main() {
    let points = [
        // outer triangle
        builder::vertex(Point3::new(0.0, 0.0, 0.0)),
        builder::vertex(Point3::new(5.0, -5.0, 0.0)),
        builder::vertex(Point3::new(10.0, 0.0, 0.0)),
        // inner triangle
        builder::vertex(Point3::new(1.0, -0.5, 0.0)),
        builder::vertex(Point3::new(9.0, -0.5, 0.0)),
        builder::vertex(Point3::new(5.0, -3.0, 0.0)),
    ];

    let outer_wire: Wire = vec![
        builder::line(&points[0], &points[1]),
        builder::line(&points[1], &points[2]),
        builder::line(&points[2], &points[0]),
    ]
    .into();
    let inner_wire: Wire = vec![
        builder::line(&points[3], &points[4]),
        builder::line(&points[4], &points[5]),
        builder::line(&points[5], &points[3]),
    ]
    .into();

    let face = builder::try_attach_plane(&vec![outer_wire, inner_wire]).unwrap(); // I've also tried face.add_boundary(inner_wire)
    let solid = builder::tsweep(&face, Vector3::unit_z());

    let mesh = solid.triangulation(0.05).to_polygon();

    let mut out = Vec::with_capacity(1024);
    stl::write(&mesh, &mut out, stl::STLType::Binary).unwrap();
    std::fs::write("output.stl", &out).unwrap();
}
```

twitchyliquid64, 2024-01-27:
> First, looking at the code you gave me, the outer triangle is not in the inner triangle. This could be a simple mistake, since this is meant to be a rough code.

My apologies, I wrote it to illustrate the issue but did not check it 100%. My bad.

> As a more essential matter, the orientation of the boundary must be strictly specified. The outer wires should be specified counterclockwise and the inner wires clockwise. The order in which the [...]

Ah! this makes a lot of sense. When you say counterclockwise / clockwise, what is that relative to? Is that orientation in the plane of the face (in which case the top and bottom faces would have edges progressing in opposite directions when looking down)?

ytanimura, 2024-01-29:
We are considering the orientation of boundaries with respect to the orientation of the faces. The `builder::try_attach_plane` sets the orientation of the plane so that the outermost boundary is counterclockwise on the plane.

twitchyliquid64, 2024-02-01:
Understood, thanks! I've managed to get truck [fully working](https://github.com/twitchyliquid64/liquid-cad/blob/main/drawing/src/l/three_d.rs#L81) in liquid-cad for extruding paths/cut-outs, as well as additional extrusions and bores.

Thank you for your kind help!

commit d7b2d891 (2021-01-30): Merge branch '55-add-comments-to-app-rs' into 'master'

add comments

Closes #55

See merge request ricos/truck/truck!71
closed 2024-02-01 by twitchyliquid64