ricosjp/truck #57: Shapeops OR function not behaving as expected when solids touch but do not overlap

MattFerraro, 2024-02-06:
Here is a minimal example:

```rust
use truck_modeling::builder::{tsweep, vertex};
use truck_modeling::{Point3, Vector3};
use truck_shapeops::or;

fn main() {
    let point_a = vertex(Point3::new(0.0, 0.0, 0.0));
    let line_a = tsweep(&point_a, Vector3::unit_x());
    let square_a = tsweep(&line_a, Vector3::unit_y());
    let cube_a = tsweep(&square_a, Vector3::unit_z());

    let z_offset = 1.0;
    let x_offset = 0.1;
    let y_offset = 0.1;
    let point_b = vertex(Point3::new(x_offset, y_offset, z_offset));
    let line_b = tsweep(&point_b, Vector3::unit_x());
    let square_b = tsweep(&line_b, Vector3::unit_y());
    let cube_b = tsweep(&square_b, Vector3::unit_z());

    let combined_cube = or(&cube_a, &cube_b, 0.01);
    match combined_cube {
        Some(c) => {
            println!(
                "combined_cube has {:?} shell boundaries",
                c.boundaries().len()
            );
        }
        None => {
            println!("combined_cube: None");
        }
    }
}

```

```bash
combined_cube: None
```

I expected the output to be:
```
combined_cube has 1 shell boundaries
```

If I change the value to: `let z_offset = 0.9` then I get the correct behavior:
```bash
combined_cube has 1 shell boundaries
``` 

If I change the value to: `let z_offset = 1.1` then I get different behavior which is also correct:
```bash
combined_cube has 2 shell boundaries
```

It is only when I choose `let z_offset = 1.0` when the None value is returned. Is this the intended behavior? How can I merge two solids which are touching but not overlapping?

MattFerraro, 2024-02-12:
Maybe I'm misunderstanding the purpose of the `or` function. Is there some other way to join two cubes together that share a face?

tsukimizake, 2025-04-29:
I tried to debug the boolean operations with coplanar faces on https://github.com/tsukimizake/truck/tree/fix-coplanar-boolean-operations-2 . Now I gave up fixing it, but at least I could make or operation of adjacent cubes work.
I'll write up what I did. I hope it helps him fix it.

I can use english to some extent, but as the author of this project seems to be also Japanese, I would like to write it in Japanese. If there is anybody interested, please use a llm to translate it.

最も単純なテストケースとして以下のようなz=1で接し、xy軸をずらすことで他の面は同一平面上にない2つのboxのorで実験しました。

```rust
#[test]
fn adjacent_cubes_or() {
    let v = builder::vertex(Point3::origin());
    let e = builder::tsweep(&v, Vector3::unit_x());
    let f = builder::tsweep(&e, Vector3::unit_y());
    let cube: Solid = builder::tsweep(&f, Vector3::unit_z());

    let v = builder::vertex(Point3::new(0.5, 0.5, 1.0));
    let w = builder::tsweep(&v, Vector3::unit_x());
    let f = builder::tsweep(&w, Vector3::unit_y());
    let cube2: Solid = builder::tsweep(&f, Vector3::unit_z());

    let result = crate::or(&cube, &cube2, 0.05);
    match &result {
        Some(_) => eprintln!("Result is Some"),
        None => eprintln!("Result is None"),
    }

    assert!(
        result.is_some(),
        "Boolean OR of adjacent cubes should succeed"
    );
}
```

これを実行したときまず失敗していたのはdivide_facesの
```rust
        let op = pre_faces.iter_mut().find(|face| face[0].poly.include(pt))?;
```
がNoneを返すというものでした。

この原因はcreate_loops_storeにおいて上のケースでのz=1のような共有面の隣で、相手Shellの辺が交線と見なされてLoopsStoreに追加されてしまうというものでした。
追加したadjacent_cubes_or testにおいては以下のような形のLoopsStoreが発生していたので、finalize_adjacent_to_coplanar_facesで`(0.5, 0.5, 1) -> (1, 0.5, 1)` と `(1, 0.5, 1) -> (0.5, 0.5, 1)`のような自明なループを削除するというパッチアップをまずは試しています。

```
LoopsStore[1]:
  Loop[0] with status: Or
  Edges in loop:
  Edge: (0.5, 0.5, 1) -> (1, 0.5, 1)
  Edge: (1, 0.5, 1) -> (1.5, 0.5, 1)
  Edge: (1.5, 0.5, 1) -> (1.5, 0.5, 2)
  Edge: (1.5, 0.5, 2) -> (0.5, 0.5, 2)
  Edge: (0.5, 0.5, 2) -> (0.5, 0.5, 1)
  Loop[1] with status: And
  Edges in loop:
  Edge: (0.5, 0.5, 1) -> (1, 0.5, 1)
  Edge: (1, 0.5, 1) -> (0.5, 0.5, 1)

```

ただ、この対処は全く完全ではなく、例えばcoplanar_faces_or testでは以下のようなLoopsStoreが発生してwire is not simpleとなります。
```rust
LoopsStore[4]:
  Loop[0] with status: Or
  Edges in loop:
  Edge: (0, 0.5, 0) -> (0, 0.5, 1)
  Edge: (0, 0.5, 1) -> (0, 0, 1)
  Edge: (0, 0, 1) -> (0, 0, 0)
  Edge: (0, 0, 0) -> (0, 0.5, 0)
  Edge: (0, 0.5, 0) -> (0, 1, 0)
  Edge: (0, 1, 0) -> (0, 1, 1)
  Edge: (0, 1, 1) -> (0, 0.5, 1)
  Edge: (0, 0.5, 1) -> (0, 0.5, 0)
  Edge: (0, 0.5, 0) -> (0, 0, 0)
  Edge: (0, 0, 0) -> (0, 0.5, 0)
```

また、現状のfinalize_adjacent_to_coplanar_facesでは共有面がカーブしている場合にも捕捉できないと思われます。

add_edgeによる面の分割やShapeOpsStatus選択のロジックが読み解ききれず手を付けられていませんが、そのあたりで面上に辺があるケースを弾く方が正道かもしれません。

ovo-Tim, 2025-05-11:
Trying to solve this, I followed what @tsukimizake said(translated):
> The root cause was found in create_loops_store, where in cases like the shared face at z=1, edges from the other shell were being mistakenly recognized as intersection curves and added to the LoopsStore.

So I wrote this function to prevent edges from the coplanar face being recognized as intersection curves and added to the LoopsStore.
``` rust
fn is_coplanar<C>(face0: &C, face1: &C) -> bool
    where
        C: ParametricSurface3D + Clone,
    {

    let n0 = face0.normal(0.0, 0.0).normalize();
    let n1  = face1.normal(0.0, 0.0).normalize();

    if !(n0.dot(n1).abs() - 1.0).so_small() {
        return false;
    }

    let pt0 = face0.subs(0.0, 0.0);
    let pt1 = face1.subs(0.0, 0.0);
    if !(pt0 - pt1).dot(n0).abs().so_small() {
        return false;
    }
    true
}
```

[image]

The change doesn’t seem to break anything (it passes all the tests), but the issue is still there. 😿

ovo-Tim, 2025-05-18:
It seems the wrong face was sent to `divide_one_face`.
[image]

```
Face Face { boundaries: [Wire { edge_list: [Edge { vertices: (Vertex { point: Mutex { data: Point3 [1.0, 0.0, 0.0] } }, Vertex { point: Mutex { data: Point3 [1.0, 1.0, 0.0] } }), orientation: true, curve: Mutex { data: FirstType(Line(Line(Point3 [1.0, 0.0, 0.0], Point3 [1.0, 1.0, 0.0]))) } }, Edge { vertices: (Vertex { point: Mutex { data: Point3 [1.0, 1.0, 0.0] } }, Vertex { point: Mutex { data: Point3 [1.0, 1.0, 1.0] } }), orientation: true, curve: Mutex { data: FirstType(Line(Line(Point3 [1.0, 1.0, 0.0], Point3 [1.0, 1.0, 1.0]))) } }, Edge { vertices: (Vertex { point: Mutex { data: Point3 [1.0, 0.0, 1.0] } }, Vertex { point: Mutex { data: Point3 [1.0, 1.0, 1.0] } }), orientation: false, curve: Mutex { data: FirstType(Line(Line(Point3 [1.0, 0.0, 1.0], Point3 [1.0, 1.0, 1.0]))) } }, Edge { vertices: (Vertex { point: Mutex { data: Point3 [1.0, 0.0, 0.0] } }, Vertex { point: Mutex { data: Point3 [1.0, 0.0, 1.0] } }), orientation: false, curve: Mutex { data: FirstType(Line(Line(Point3 [1.0, 0.0, 0.0], Point3 [1.0, 0.0, 1.0]))) } }] }], orientation: true, surface: Mutex { data: Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) } }
```

ovo-Tim, 2025-05-24:
There seems to be an issue in the `intersection_curve::intersection_curves` function, which fails to detect the intersection curve when two planes are coincident — specifically, when both planes are defined as `Z=0`.

<details>
<summary>The output of  intersection_curves</summary>

```
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [1.5, 0.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [1.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [0.5, 1.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [0.5, 1.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 1.0, 0.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 2.0], p: Point3 [1.5, 0.5, 2.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [1.5, 0.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [1.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 1.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [0.5, 1.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 0.0, 0.0], p: Point3 [1.0, 0.0, 0.0], q: Point3 [0.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 2.0], p: Point3 [1.5, 0.5, 2.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] })
Intersection Curve: IntersectionCurveWithParameters { ic: IntersectionCurve { surface0: Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }), surface1: Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] }), leader: PolylineCurve([Point3 [1.0, 0.5, 1.0], Point3 [1.0, 1.0, 1.0]]) }, params0: PolylineCurve([Point2 [0.5, 1.0], Point2 [1.0, 1.0]]), params1: PolylineCurve([Point2 [0.5, 0.0], Point2 [0.5, 0.5]]) }
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [1.5, 0.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [1.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 1.5, 1.0], p: Point3 [1.5, 1.5, 1.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [0.5, 1.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
Face Plane(Plane { o: Point3 [1.0, 0.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [1.0, 0.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 2.0], p: Point3 [1.5, 0.5, 2.0], q: Point3 [0.5, 1.5, 2.0] })
Face Plane(Plane { o: Point3 [0.0, 1.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [0.0, 1.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] })
Intersection Curve: IntersectionCurveWithParameters { ic: IntersectionCurve { surface0: Plane(Plane { o: Point3 [0.0, 1.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [0.0, 1.0, 1.0] }), surface1: Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 1.5, 1.0] }), leader: PolylineCurve([Point3 [0.5, 1.0, 1.0], Point3 [1.0, 1.0, 1.0]]) }, params0: PolylineCurve([Point2 [0.5, 1.0], Point2 [1.0, 1.0]]), params1: PolylineCurve([Point2 [0.0, 0.5], Point2 [0.5, 0.5]]) }
Face Plane(Plane { o: Point3 [0.0, 1.0, 0.0], p: Point3 [1.0, 1.0, 0.0], q: Point3 [0.0, 1.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 1.0], p: Point3 [1.5, 0.5, 1.0], q: Point3 [0.5, 0.5, 2.0] })
[... cut ...] 0.5, 0.5, 1.0], p: Point3 [0.5, 1.5, 1.0], q: Point3 [0.5, 0.5, 2.0] }), leader: PolylineCurve([Point3 [0.5, 0.5, 1.0], Point3 [0.5, 1.0, 1.0]]) }, params0: PolylineCurve([Point2 [0.5, 0.5], Point2 [0.5, 1.0]]), params1: PolylineCurve([Point2 [0.0, 0.0], Point2 [0.5, 0.0]]) }
Face Plane(Plane { o: Point3 [0.0, 0.0, 1.0], p: Point3 [1.0, 0.0, 1.0], q: Point3 [0.0, 1.0, 1.0] }) and Face Plane(Plane { o: Point3 [0.5, 0.5, 2.0], p: Point3 [1.5, 0.5, 2.0], q: Point3 [0.5, 1.5, 2.0] })
```

</details>

ovo-Tim, 2025-05-24:
Since `intersection_curve::intersection_curves` only detects surface intersections, we may need a specific algorithm to handle coplanar faces.

ovo-Tim, 2025-05-25:
I tried to manually add lines to geom_loops when it's coplanar for testing. However, I failed to satisfy some traits.

`truck-shapeops/src/transversal/loops_store/mod.rs`
``` rust
fn is_coplanar<S>(surface1: &S, surface2: &S) -> bool
where
    S: ParametricSurface3D + SearchNearestParameter<D2, Point = Point3>,
{
    let norm1 = surface1.normal(0.0, 0.0).normalize();
    let norm2 = surface2.normal(0.0, 0.0).normalize();
    if !(norm1.dot(norm2) - 1.0).so_small() {
        return false;
    }
    if !(surface1.subs(0.0, 0.0) - surface2.subs(0.0, 0.0)).dot(norm1).so_small() {
        return false;
    }
    true
}
```

`truck-shapeops/src/transversal/loops_store/mod.rs create_loops_stores`
``` rust
if is_coplanar(&surface0, &surface1) {

                // println!("Coplanar surfaces:  {:?} {:?}", surface0, surface1);
                let v1 = Vertex::new(Point3::new(0.5, 0.5, 1.0));
                let v2 = Vertex::new(Point3::new(0.5, 1.0, 1.0));
                let v3 = Vertex::new(Point3::new(1.0, 0.5, 1.0));
                let v4 = Vertex::new(Point3::new(1.0, 1.0, 1.0));
                let curve1 = PolylineCurve::from(vec![v1.point(), v2.point()]);
                let curve2 = PolylineCurve::from(vec![v1.point(), v3.point()]);
                let curve3 = PolylineCurve::from(vec![v2.point(), v4.point()]);
                let curve4 = PolylineCurve::from(vec![v3.point(), v4.point()]);

                let edge1 = Edge::new(&v1, &v2, curve1.into());
                let edge2 = Edge::new(&v1, &v3, curve2.into());
                let edge3 = Edge::new(&v2, &v4, curve3.into());
                let edge4 = Edge::new(&v3, &v4, curve4.into());

                geom_loops_store0[face_index0].add_edge(edge1.clone(), ShapesOpStatus::Or);
                geom_loops_store1[face_index1].add_edge(edge1, ShapesOpStatus::Or);
                geom_loops_store0[face_index0].add_edge(edge2.clone(), ShapesOpStatus::Or);
                geom_loops_store1[face_index1].add_edge(edge2, ShapesOpStatus::Or);
                geom_loops_store0[face_index0].add_edge(edge3.clone(), ShapesOpStatus::Or);
                geom_loops_store1[face_index1].add_edge(edge3, ShapesOpStatus::Or);
                geom_loops_store0[face_index0].add_edge(edge4.clone(), ShapesOpStatus::Or);
                geom_loops_store1[face_index1].add_edge(edge4, ShapesOpStatus::Or);
            }
```

ovo-Tim, 2025-05-28:
Based on @tsukimizake 's code, I likely managed to clean all the "trivial edges". However, test `coplanar_faces_or` still fails with the error: "wire is not simple".

[image]

``` rust
fn troublesome_list<C>(wires: &BoundaryWire<Point3, C>, ifront: usize, iback: usize) -> Vec<usize> {
    let mut troublesome = vec![];
    if wires[ifront].front() == wires[iback].back(){
        troublesome.push(ifront);
        troublesome.push(iback);
        if ifront > 0 && iback < wires.len() - 1{
            troublesome.extend(troublesome_list(wires, ifront-1, iback+1));
        }
    }
    troublesome
}

fn finalize_adjacent_to_coplanar_faces<C, S>(
    geom_loops_store: &mut LoopsStore<Point3, C>,
    face_index: usize,
) -> ()
where
    S: ParametricSurface3D + SearchNearestParameter<D2, Point = Point3> + Debug,
    C: Debug,
{
    let mut remove_loops = vec![];
    geom_loops_store[face_index]
        .iter_mut()
        .enumerate()
        .for_each(|(iloop, loop_)| {
            let mut remove_edge = vec![];
            (0..loop_.len()-1).for_each(|iedge|{
                remove_edge.extend(troublesome_list(loop_, iedge, iedge+1));
            });
            remove_edge.sort_unstable();
            remove_edge.into_iter().rev().for_each(|index| {
                loop_.remove(index);
            });
            if loop_.len() == 0{
                remove_loops.push(iloop);
            }
        });
    remove_loops.into_iter().rev().for_each(|index| {
        geom_loops_store[face_index].remove(index);
    });
}
```

joeblew999, 2026-04-05:
I'm a downstream user building on truck — sharing context from the [monstertruck](https://github.com/virtualritz/monstertruck) fork.

This is a specific instance of the broader coincident/touching geometry problem (also see #114). The root issue is that when two solids share an exact face (`z_offset = 1.0`), the transversal intersection code can't find a valid intersection curve — the faces are coplanar rather than transversally intersecting — so it returns `None`.

The [monstertruck](https://github.com/virtualritz/monstertruck) fork changes boolean ops to return `Result<Solid, ShapeOpsError>` instead of `Option<Solid>` (see [`monstertruck-solid/src/lib.rs L20`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-solid/src/lib.rs#L20)), which at minimum makes the failure diagnosable rather than silent — you'd get `Err(ShapeOpsError::CoincidentFaces)` or similar rather than a bare `None`. It doesn't fully fix the coplanar case but makes the failure mode actionable.

The proper fix — handling coplanar/touching faces as a special case in the transversal intersection — is a harder algorithmic problem. The upstream lib.rs doc comment acknowledges this: "Cases where faces are tangent to each other are not yet supported."

commit 13e29a52 (2021-01-31): Merge branch '57-mapped-shell' into 'master'

Resolve "mapped の Shell 移動がおかしい"

Closes #57

See merge request ricos/truck/truck!74