ricosjp/truck #67: Integration Into CADmium

MattFerraro, 2024-05-27:
Hello,

I just wanted to reach out as a courtesy and say that I wrote a big article about the future of CAD programs. You can read it [here](https://mattferraro.dev/posts/cadmium). This article got some attention on [Hacker News](https://news.ycombinator.com/item?id=40428827) and also [Hackaday](https://hackaday.com/2024/05/23/cadmium-moves-cad-to-the-browser/). All that attention allowed us to get together more than 1000 interested individuals in a [Discord Server](https://discord.gg/qJCsKJeyZv) to work together to implement a [new CAD program](https://cadmium-co.github.io/CADmium/) [built on top of Truck](https://github.com/CADmium-Co/CADmium).

To be successful, we will probably need to add new features to Truck. We would love to work together to figure out how we can collaborate and push any changes we make back upstream to you.

We are very grateful that you are building Truck and we see it as a foundational technology for the future of manufacturing. I am available at mattferraro.dev@gmail.com if you would like to talk.

Thank you,
Matt Ferraro

ytanimura, 2024-05-28:
Dear Matt Ferraro,

Thank you very much for the very good news. Your project is being talked about tremendously and has even been featured in a technology article in Japan.

I read your article on the future vision and it is very well researched on mechanical CAD. The idea of a Git-like management of the design process was also a concept that was in the back of my mind when I first started developing `truck`, but I gave up on it because it was too difficult for me to implement due to my resources. If you can make it happen, then all the better.

As the article says, `truck` is still young and immature. The most immediate challenge for incorporating it into a practical CAD system will be to speed up and stabilize the boolean operations. As you can see from the commit log, not much resources have been devoted to the development of `truck` itself recently. This is because we are devoting more man-hours to `truck` operations. We are developing in a more direct manner in order to achieve our main goal, which is to optimize shapes through automatic geometry generation by `truck` and repeated CAE accelerated by machine learning. However, in order to scale our business in the future, it is essential to improve the performance of `truck` itself, and we are still concerned about devoting resources to boolean operation. Any help with the development of `truck` itself would be greatly appreciated.

We also believe that an open, browser-based CAD system would be very useful for our platform. We hope to achieve a mutually beneficial relationship in the future.

I have replied in haste, but I should contact you by e-mail after talking with my superiors regarding specific details. Thank you again for your wonderful project!

Yoshinori Tanimura

MattFerraro, 2024-05-29:
Excellent to hear from you! I look forward to chatting more!

As an immediate next step, I'll start slowly filing issues in the truck repo and seeing if anyone in our team is able to help fix them and push PR's back to your team.

Cheers,
Matt

commit 111a37e5 (2021-02-20): Merge branch '67-shapeinstance-oneobject' into 'master'

Resolve "ShapeInstance をひとつに纏める"

Closes #67

See merge request ricos/truck/truck!94