ricosjp/truck #73: Can truck load stp, step models and render them? Which library do I need?

vltown, 2024-07-01:
Can truck load stp, step models and render them? Which library do I need?

ytanimura, 2024-07-02:
There are some sample codes.

Convert Step -> Mesh
https://github.com/ricosjp/truck/blob/master/truck-stepio/examples/step-to-mesh.rs

Render mesh
https://github.com/ricosjp/truck/blob/master/truck-rendimpl/examples/simple-obj-viewer.rs

vltown, 2024-07-02:
> https://github.com/ricosjp/truck/blob/master/truck-rendimpl/examples/simple-obj-viewer.rs

I have seen this, but this method directly converts the json file generated by truck into an obj file, and then renders the obj file to implement the rendering model. Is this the only way?

ytanimura, 2024-07-02:
I think what you say is [simple-shape-viewer](https://github.com/ricosjp/truck/blob/master/truck-rendimpl/examples/simple-shape-viewer.rs).
simple-obj-viewer renders the obj file directly. One can convert step to obj by step-to-mesh. In fact, it is sufficient to use `PolygonMesh`, which is an intermediate structure, without actually spitting it out into an obj file. Anyway, it is necessary to convert a shape in a step file to a mesh.

vltown, 2024-07-05:
> I think what you say is [simple-shape-viewer](https://github.com/ricosjp/truck/blob/master/truck-rendimpl/examples/simple-shape-viewer.rs).我认为你说的是​​简单形状查看器。 simple-obj-viewer renders the obj file directly. One can convert step to obj by step-to-mesh. In fact, it is sufficient to use `PolygonMesh`, which is an intermediate structure, without actually spitting it out into an obj file. Anyway, it is necessary to convert a shape in a step file to a mesh. simple-obj-viewer 直接渲染 obj 文件。可以通过step-to-mesh将step转换为obj。事实上，使用 `PolygonMesh` 就足够了，它是一个中间结构，而不需要实际将其吐出到 obj 文件中。无论如何，有必要将步骤文件中的形状转换为网格。

How should polyshells be used? Is polyshells related to truck_modeling::topology::Shell?

ytanimura, 2024-07-10:
Perhaps what you are calling "polyshell" is a shell with the mesh as geometry. The process of shell and solid meshing is:

shell with parametric geometry --(`robust_triangulation`)--> shell with polygonmesh geometry --(`to_polygon`)--> polygonmesh.

When performing processes such as color-coding each face or acquiring the picked face on the display, it is necessary to remember the correspondence between the mesh and the topology, so a "polyshell" is generated in the interim. If you just want to draw a shape, you can combine the entire mesh data into one. The method to_polygon can do that.

vltown, 2024-07-10:
> Perhaps what you are calling "polyshell" is a shell with the mesh as geometry. The process of shell and solid meshing is:
>
> shell with parametric geometry --(`robust_triangulation`)--> shell with [...]

I have a problem now: I want to use Rust's wgpu to render the step model. I use truck to parse the model, but there are always problems with the vertices and indices I get. Is there any example I can refer to?

ytanimura, 2024-07-10:
I do not know what exactly you are having trouble with, but here is the code in the `truck` that registers the polygons to a buffer on `wgpu`.

https://github.com/ricosjp/truck/blob/master/truck-rendimpl/src/polyrend.rs#L4

vltown, 2024-07-12:
> I do not know what exactly you are having trouble with, but here is the code in the `truck` that registers the polygons to a buffer on `wgpu`.
> 
> https://github.com/ricosjp/truck/blob/master/truck-rendimpl/src/polyrend.rs#L4

thanks

closed 2024-07-12 by vltown