ricosjp/truck #74: How to solve NotSimpleWire?

vltown, 2024-07-12:
`let shells: Vec<truck_topology::Shell<Point3, Curve3D, Surface>> = table.shell.values().cloned().filter_map(|step_shell| {
        let mut cshell = match table.to_compressed_shell(&step_shell) {
            Ok(shell) => shell,
            Err(e) => {
                println!("Error during compression: {:?}", e);
                return None; // 跳过这个shell处理下一个
            }
        };
        cshell.robust_split_closed_edges_and_faces(0.1);
        match truck_topology::Shell::extract(cshell) {
            Ok(shell) => Some(shell),
            Err(e) => {
                println!("Failed to extract shell: {:?}", e);
                None // 处理失败时返回None, 继续处理其他数据
            }
        }
    }).collect();`
`Failed to extract shell: NotSimpleWire`
This code will cause this error, causing my model to render incompletely. Is there any way to solve this problem?

ytanimura, 2024-07-16:
Perhaps your STEP data does not meet the requirements. Boundary representation in `truck` does not accept boundary wires with self-intersections. This is a problem when editing the shape, and if you just want to display it, just leave the meshing as `CompressedShell`. If you want to edit the shape, try this method.

https://gitlab.ritc.jp/ricos/truck/truck/-/blob/master/truck-shapeops/src/healing/mod.rs?ref_type=heads#L80

joeblew999, 2026-04-05:
I'm a downstream user building on truck — sharing a reference from the [monstertruck](https://github.com/virtualritz/monstertruck) fork.

The `NotSimpleWire` error after `robust_split_closed_edges_and_faces` usually means the healed shell still has topological issues that prevent clean extraction. The fork adds a dedicated [`extract_healed`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-solid/src/healing/mod.rs#L136) function in `monstertruck-solid` that combines the split + extract into a single pipeline with better error recovery:

```rust
pub fn extract_healed<C, S>(
    shell: CompressedShell<Point3, C, S>,
    tolerance: f64,
) -> Result<Shell<Point3, C, S>, Error>
```

Rather than calling `robust_split_closed_edges_and_faces` then `Shell::extract` separately (where the extract can still fail), `extract_healed` applies additional healing passes to resolve wires that remain non-simple after splitting. It's exported from [`monstertruck-solid/src/lib.rs`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-solid/src/lib.rs).

Apache-2.0 — may be useful as a reference or for porting to upstream `truck-shapeops`.

commit 5eccb7a8 (2021-02-24): Merge branch '74-renderid-atomicusize' into 'master'

Resolve "RenderID の実体を AtomicUsize にする"

Closes #74

See merge request ricos/truck/truck!98