ricosjp/truck #79: Is it possible to create a helix? 🧬 (modeling screws)

twitchyliquid64, 2024-10-25:
For modeling threads, it would be great to ‘rsweep’ the thread profile (trapezoidal or V shaped) in such a way that it rotates multiple times and gains height, then combine that with a cylinder to form a screw.

[image]

Is a helical rsweep possible today in truck? If not so, may I kindly ask a feature request?

ytanimura, 2024-10-25:
Well, we have a plan, but we are still struggling with it. In the end, we have to make a decision somewhere.
Implementing a spiral structure with NURBS results in an approximation, so we should create a specific structure for spiral, but it cannot be directly output that structure to STEP.

twitchyliquid64, 2024-10-28:
Understood, thank you for the reply!

I wonder if a naive implementation could be a specific structure which, as the u/v parameter along the spiral increases, the returned curve/vector (in ‘ParametricCurve’ methods) translates proportionally in the direction of the spiral. I do not understand all of the traits to know where this won’t work however.

hyarion, 2025-01-20:
Worth noting is that OCCT struggles with this too, see https://opencascade.blogspot.com/2015/08/arbitrary-law-based-curve-and-surface.html for workaround there.
It might be good to support non-standard curves and surfaces and do the approximation at export instead.

IIRC other (proprietary) kernels that support helices handles it in that ways - handle helices exact internally and do approximations when needed.

commit b1b33f63 (2021-03-03): Merge branch '79-wireframe-from-polygonmesh' into 'master'

Resolve "WireFrame from PolygonMesh"

Closes #79

See merge request ricos/truck/truck!104