ricosjp/truck #85: Provide `difference` (`and_not`) and `symmetric_difference` (`xor`) operation

uzytkownik, 2024-12-08:
I figure out I need to `.not` shape by finding issue #68. It would be nice if truck provided .not() operation even if for now they would be implemented in terms of `.or` and `.and`.

ytanimura, 2024-12-16:
In boundary expressions, if you turn over all the faces of the shell, it's the same as doing a not operation. That's why I decided not to add it here, because I thought it wasn't something that should be listed in parallel. I might think about it again when it's a bit more complete.

ytanimura, 2024-12-19:
Well, I'm sorry to say that I forgot, but the method `not` was defined directly in `Solid`.
https://docs.rs/truck-topology/0.6.0/truck_topology/struct.Solid.html#method.not

uzytkownik, 2025-02-02:
I think it's most common for CADs to have difference and symmetric difference operations than negation. So I think many people will search for difference rather than not operation at first.

joeblew999, 2026-04-05:
I'm a downstream user building on truck — sharing a reference from the [monstertruck](https://github.com/virtualritz/monstertruck) fork.

Agreeing with @uzytkownik above — `difference` is the term people reach for first, and `not` + manual combination isn't obvious from the docs.

The fork implements both missing operators in [`monstertruck-solid/src/lib.rs L20`](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-solid/src/lib.rs#L20):

```rust
pub fn difference(solid0: &Solid, solid1: &Solid, tol: f64) -> Result<Solid, ShapeOpsError>
pub fn symmetric_difference(solid0: &Solid, solid1: &Solid, tol: f64) -> Result<Solid, ShapeOpsError>
```

Both follow the same transversal intersection approach as `and`/`or` and are exposed in the [WASM bindings](https://github.com/virtualritz/monstertruck/blob/682c67622b4a04863fc4151c63ddcd07c6b4bb69/monstertruck-wasm/src/shapeops.rs) too.

Note these return `Result<Solid, ShapeOpsError>` rather than `Option` — which also helps with the silent failure problem discussed in #91 and #114. Apache-2.0 if useful as a reference.