ricosjp/truck #87: Please open the discussion forum for this repository

kcleung, 2025-01-31:
It would be great if we can open the discussion forum for this github repository, so that ideas can be discussed there, instead of unnecessarily opening an issue just for the purpose of asking questions, discussing ideas,  possible implementations or future directions.

devrangel, 2025-02-02:
A Discord channel would be great!

kcleung, 2025-02-03:
> A Discord channel would be great!

But it would be hard to search for discord coversations.  The whole point for a forum is that form can be organized into folders and topics, and all discussions are easily searchable

ytanimura, 2025-02-06:
If there are any examples of projects that are doing something like the discussion forum issue, I would like to use them as a reference.

tymokvo, 2025-02-26:
Just FYI, [`nalgebra`](https://github.com/dimforge/nalgebra/discussions) and [`bevy`](https://github.com/bevyengine/bevy/discussions) also use GitHub Discussions in this way. [Fornjot](https://www.fornjot.app/) has a [Matrix](https://matrix.to/#/#fornjot:pub.solar).

j209, 2025-03-18:
fornjot uses GitHub Discussions too.