ricosjp/truck #89: 120GB Adjacency Matrix when running `PolygonMesh::put_together_same_attrs`

GamingLiamStudios, 2025-02-05:
it literally crashes my computer all i want is a triangle mesh lol.

Jokes aside, I'm getting this issue when I'm optimizing the triangulization of a step file with 176241 IDs, or 121361 verts (i think). This specifically only happens with the final attribute set of the Normals. 
Github won't let me upload the .obj file of the mesh before the optim step, so i'll figure out how to get that here.

GamingLiamStudios, 2025-02-05:
oh and here's my code
```rust
use std::error::Error;

use truck_meshalgo::prelude::*;
use truck_stepio::r#in::*;
use truck_topology::compress::*;

pub fn main() -> Result<(), Box<dyn Error>> {
    let file = std::fs::read_to_string("test.step")?;

    let mut polymesh = PolygonMesh::default();

    let table = Table::from_step(&file).expect("shit");
    for shell in table.shell.values() {
        println!("First Pass");
        let shell = table.to_compressed_shell(shell)?;
        let mut poly = shell.robust_triangulation(0.01).to_polygon();
        println!("Optimize");
        poly.put_together_same_attrs(TOLERANCE * 50.0)
            .remove_degenerate_faces()
            .remove_unused_attrs();
        polymesh.merge(poly);
    }

    obj::write(&polymesh, std::fs::File::create("test.obj")?)?;
    Ok(())
}
```

GamingLiamStudios, 2025-02-05:
Looking at the code responsible only confuses me. What unit of measurement is 'tolerance' in? What is the purpose of culling the UV and Normal attribute streams as well?

ytanimura, 2025-02-06:
Thank you for your comment.

I cannot say for sure because I do not know the error message, but I think there are NAN values in some normals because some degenerate surfaces are included in the STEP file. For example, try adding a process to remove NANs as follows.

```rust
use std::error::Error;

use truck_meshalgo::prelude::*;
use truck_stepio::r#in::*;

// ---- added
fn nan2zero(x: &mut f64) {
    if x.is_nan() {
        *x = 0.0;
    }
}
// ----

pub fn main() -> Result<(), Box<dyn Error>> {
    let file = std::fs::read_to_string("test.step")?;

    let mut polymesh = PolygonMesh::default();

    let table = Table::from_step(&file).expect("shit");
    for shell in table.shell.values() {
        println!("First Pass");
        let shell = table.to_compressed_shell(shell)?;
        let mut poly = shell.robust_triangulation(0.01).to_polygon();
        // ---- added
        poly.normals_mut().iter_mut().for_each(|n| {
            nan2zero(&mut n.x);
            nan2zero(&mut n.y);
            nan2zero(&mut n.z);
        });
        // ----
        println!("Optimize");
        poly.put_together_same_attrs(TOLERANCE * 50.0)
            .remove_degenerate_faces()
            .remove_unused_attrs();
        polymesh.merge(poly);
    }

    obj::write(&polymesh, std::fs::File::create("test.obj")?)?;
    Ok(())
}
```

GamingLiamStudios, 2025-02-06:
Unfortunately still no work :(
anyway here's [step](https://litter.catbox.moe/7zkemb.step) so you guys can have a crack. Just hosted on a temp file server i know, should be there for like 3 days or so.

ytanimura, 2025-02-18:
It will take some time to execute, but at least the memory explosion should be resolved.

commit e541ff5d (2021-05-24): Merge branch '89-tessellation' into 'master'

Resolve "Tessellation"

Closes #89

テスト実装は別イシューで行う。

See merge request ricos/truck/truck!117