ricosjp/truck #93: Pattern for implementing geometry with shared BRep and unique transformations

tymokvo, 2025-02-26:
Hello,

I am using truck to analyze geometry from IFC files. Similar to STEP, IFC can represent the concept of instancing where one BRep is transformed into multiple locations in a model to represent multiple entities. 

Is there a recommended way to implement this using truck?

Some options that I have considered:
- During all `truck_modeling::builder` operations, thread `Matrix4` through each stage, transforming every `Point3`/`Vector3` that is used to create the shape
- After all `truck_modeling::builder` operations, use `PolygonMesh::{positions_mut, normals_mut}` to mutate the polygon vertices/normals of the resulting mesh for display

ytanimura, 2025-02-28:
This is not a builder, but rather something that should be implemented at a higher level of assembly. It is a graph structure with nodes that perform shape data or matrix conversion, and it is used to represent the assembly of products and the interior of model rooms. The implementation within `truck` has been roughly conceptualized, but it will be implemented once the investigation into its representation in STEP format is complete.

tymokvo, 2025-02-28:
Ok, thank you! For now, I have used the second option above, only applying the transformation to the generated mesh and it is working for now. 

If there is any way to help out it would be nice to contribute. I will close for now though.

commit 6fe3a93f (2021-10-03): Merge branch '93-add-explanation-of-tolerance-limits-and-error-handling' into 'master'

Resolve "Add explanation of tolerance limits and error handling"

Closes #93

See merge request ricos/truck/truck!153